jmamma/MegaCommand_Design #5: Alternative Chassis Design (migrated)

yatli, 2019-08-24:
I'm working on a metal chassis design.

Basically, I'd like to make it low-cost, durable and mountable. The current design consists of two parts, each can be produced with sheetmetal (304, 6061 etc.)

[image]

Potential problems:

The screw tap hole "ears" are less than optimal. May consider bend the whole edge instead of cutting those 6 ears.
Screen height?
No power switch hole...

yatli, 2019-08-24:
The first batch has arrived. Lots of bugs, but I managed to fit it in:

[image]

[image]

[image]

yatli, 2019-08-24:
== Problems ==

- [ ] The OLED Display window misaligns. It should be ~1mm left to the current position.
- [ ] The OLED Display window should be higher. Currently it mates very tightly with the metal case of the display.
- [ ] The left LED should be ~1mm left to the current position.
- [ ] Encoder through holes should be larger.
- [ ] Mount points is a joke -- one of them projects right onto the usb port!
- [ ] Key switch through holes should be larger.

yatli, 2019-08-24:
The photo is different from the original design that the chassis tap drills are now all on the base plate.
The panel plate "envelopes" the base plate now.
The sheet metal flanges are also simplified to bring down the cost.
I've got 2 pairs at RMB 150 each.

yatli, 2019-08-24:
Caveats:

1. Put insulation washers between the board and the copper standoffs. Otherwise it will short some pins.

2. Cut the washers so that the profile stays inside the pcb area:

[image]

3. Tape the back of the panel around the OLED window. Otherwise it will short the display pins after assembly.

yatli, 2019-08-27:
Yeah.
Full Metal Jacket.

jmamma, 2019-08-27:
Heavy Duty.

yatli, 2019-11-08:
Having a metal chasis is a bad idea. I almost fried something in MD when the rack mount screw contact with the MC metal chasis -- MD GND is -5V to USB GND, and a short of the two results in MD voltage dropping to 4V, which is very dangerous.

closed 2019-11-08 by yatli