jmamma/MegaCommand_Design #9: sram test fails

frnktrpz, 2020-05-03:
Hi, opening this issue after asking for help over at elektronauts after running into some issues as a possible consequence of the (classic?) mistake of soldering the oled before the sd-card holder. 

Before desoldering the oled screen all test firmwares ran successfully. After desoldering the sram test fails instantly giving the following error:

>  Megank0 mismatch 
2200
MegaCommand SRAM Test
2200
SRAM Test Failed bank0 mismatch 
2200
 
Tried several times, tried reseating the board just in case but the same error occurs. 
Before replacing the ram it would be great if anyone could shed any light on other possible causes than a damaged/faulty ram chip.

Here are pictures of my build:   
[image]
[image]

jmamma, 2020-05-03:
Can you take a clearer version of the bottom side of the board. i want to see the traces around the sram chip clearly

thanks

frnktrpz, 2020-05-03:
Hope these are more useful: 

[image]
[image]

jmamma, 2020-05-03:
I get the same test error when I attempt to run the SRAM test with the arduino disconnected from the main board.

This means theres a connectivity problem somewhere. 

```
Mest
2200

Failed bank0 zero reading
2200
MegaCommand SRAM Test
2200

Failed bank0 zero reading
2200
```

1)

For sanity, reconnect the Arduino and try again. Check that all the male header pins are aligned correctly with the Arduino when attaching it.

2)

Was the SRAM test working prior to the display problems?

3)

Can you make sure that there are no short circuits between solder joints. I see a few spots where there is a little bit of solder that might be causing a short between two unrelated pads.

4)

Do you own a multimeter?

Check that the SRAM chip is getting 5V. Pin32 connected to capacitor C4.
https://docs.rs-online.com/6fa1/0900766b8139de13.pdf

and check GND.

5)

The next step would be to verify the traces between the SRAM chip and the double row header. start with Power and GND and then work through the other traces. Keep a checklist. 

--

Troubleshooting may seem daunting, but it's a digital circuit and we can number down the possibilities by working systematically.

frnktrpz, 2020-05-03:
> I get the same test error when I attempt to run the SRAM test with the arduino disconnected from the main board.
> 
> This means theres a connectivity problem somewhere.
> 
> ```
> Mest
> 2200
> 
> Failed bank0 zero reading
> 2200
> MegaCommand SRAM Test
> 2200
> 
> Failed bank0 zero reading
> 2200
> ```
> 
> 
> For sanity, reconnect the Arduino and try again. Check that all the male header pins are aligned correctly with the Arduino when attaching it.
> 
> 
> Was the SRAM test working prior to the display problems?
> 
> 
> Can you make sure that there are no short circuits between solder joints. I see a few spots where there is a little bit of solder that might be causing a short between two unrelated pads.
> 
> 
> Do you own a multimeter?
> 
> Check that the SRAM chip is getting 5V. Pin32 connected to capacitor C4.
> https://docs.rs-online.com/6fa1/0900766b8139de13.pdf
> 
> and check GND.
> 
> 
> The next step would be to verify the traces between the SRAM chip and the double row header. start with Power and GND and then work through the other traces. Keep a checklist.
> 
> --
> 
> Troubleshooting may seem daunting, but it's a digital circuit and we can number down the possibilities by working systematically.

Thank you for the amazing support, really appreciate it!
1. 
Headers appear to be aligned correctly, reseating the board doesn't seem to make a difference.
2.
Sram test was successful before desoldering the oled screen.
3.
Cleaning the board seems to have helped a bit as the test now runs for for a while before giving the error:
>2A00
SRAM Test Failed bank0 mismatch 
2A00

Test seems to fail at the same point each time. 
4.
If I measure voltage between the large GND on the MC pcb and pin 32 on the sram my multimeter indicates a short. 
5. 
I installed autodesk so I can systematically start working through the various test points. Will probably take me a while as I'm not used yet to navigating schematics.

jmamma, 2020-05-03:
I can send you a screenshot of the top and bottom board traces.

frnktrpz, 2020-05-03:
That would be great, thanks.

jmamma, 2020-05-03:
[megacommand_1_0_2a12_top_bottom.pdf](https://github.com/jmamma/MegaCommand_Design/files/4570198/megacommand_1_0_2a12_top_bottom.pdf)

Viewed from top.

RED: top traces
BLUE: bottom traces.

jmamma, 2020-05-03:
Can you measure voltage across the double header pins GND and 5V.

frnktrpz, 2020-05-03:
I get around 4.5V reading across.

jmamma, 2020-05-03:
I read 4.5V as well.

Now check GND (from double pin header) to the supply voltage of each chip

jmamma, 2020-05-03:
> "edit;
> Gave the board another clean and instantly got a bank0 read error, reconnected the usb cable and back to the initial bank0 mismatch error.
> Not sure if relevant." 

can't really make any conclusions at this stage. 

If measuring voltages doesn't reveal anything unusual. Disconnect the arduino and start looking at the traces.

It's getting late here. So i will check in the morning to see your progress.

frnktrpz, 2020-05-03:
> I read 4.5V as well.
> 
> Now check GND (from double pin header) to the supply voltage of each chip

Consistently getting around 4.7 volts across. Except for the level shifter which shows around 3.3V.

> > "edit;
> > Gave the board another clean and instantly got a bank0 read error, reconnected the usb cable and back to the initial bank0 mismatch error.
> > Not sure if relevant."
> 
> can't really make any conclusions at this stage.
> 
> If measuring voltages doesn't reveal anything unusual. Disconnect the arduino and start testing each trace. Start with connections from the double header to the SRAM chip. Then the SRAM chip to the 74HC573 octal latch.
> 
> It's getting late here. So i will check in the morning to see your progress.

Ok, thanks again! Should keep me busy for a while. Have good night.

frnktrpz, 2020-05-04:
I checked continuity for all the traces to/from the ram ic with my multimeter and was not able to find any fault. Didn't measure resistance, just checked the traces. 
All firmware tests besides the ram test (and the oled test, am unable to test now) run successfully. Is it safe to assume that the ram chip needs to be replaced or am I jumping the gun?

jmamma, 2020-05-05:
My only other suggestion would be to re-flow the solder joints on the SRAM and octal latch. This would rule out a cold solder joint...

Another less scientific test would be to apply constant physical pressure to SRAM chip (with your thumb) and re-run the SRAM test. See if the test proceeds beyond 0x2200. This may reveal an intermittent connection...

Just be weary that desoldering the SRAM will be harder than removing the OLED. the pads are much smaller, and there is a risk of stripping them from the board.

yatli, 2020-05-05:
@frnktrpz also check the bank selection pin (PIN43)

You can enable the debug information in the test firmware, around line 57: https://github.com/jmamma/MegaCommand_Design/blob/3d7930fea5477b9c5ef57f928413bf8be208d8a9/TestFirmwares/megacommand_sram_test/megacommand_sram_test.ino#L57-L60

Uncomment these, and see if bank1 reading is correct but bank0 is not.

jmamma, 2020-05-05:
@yatli If bank select is not working, then it should read/write to the same bank. 

Previously he was getting as far as 2A00

which is 2560 bytes.. I thought this may be indicative of an address line fault. But it doesn't line up with the addressing scheme.( It's been a while since I've looked at this. Maybe my interpretation is wrong.)

2^11 = 2048
2^10 = 1024
2^9 = 512

yatli, 2020-05-05:
Ah, yes.
But I still recommend printing out the values -- if we get the same wrong value from both banks it means something (I'm not sure what it is yet)

jmamma, 2020-05-05:
SRAM Test Failed bank0 mismatc

Yeah he's getting a mismatch. Which means the value read from bank 1 doesn't match the value read from bank 0. 

A static address line fault wouldn't cause this.
A static bank select  line fault wouldn't cause this.

yatli, 2020-05-05:
The test FW doesn't compare bank1 read with bank0 read -- it compares both with the source random number, and when bank0 check fails, it doesn't report what's happening on bank1, just exit.

> which is 2560 bytes.. I thought this may be indicative of an address line fault. But it doesn't line up with the addressing scheme.( It's been a while since I've looked at this. Maybe my interpretation is wrong.)
> 
> 2^11 = 2048
> 2^10 = 1024
> 2^9 = 512

Hypothesis: 2^11 and 2^9 pins shorted?

jmamma, 2020-05-05:
https://andybrown.me.uk/2011/08/28/512kb-sram-expansion-for-the-arduino-mega-design/

reference ^

jmamma, 2020-05-05:
``` When choosing a latch we must ensure that it is fast enough to cope with the 16Mhz core clock of the Arduino. 74HC series latches are too slow. The datasheet recommends that we use the 74AHC series. ```

That's interesting.

jmamma, 2020-05-05:
@yatli what type of latch are you using.

I'm running 74AHC series.

yatli, 2020-05-05:
I've got `SN74HC573N`

jmamma, 2020-05-05:
yeah I just checked a few of my boards. I have a mix of 74AH and 74H. Presumably if this was a problem it would have come up in my testing.

yatli, 2020-05-05:
I didn't know the difference between HC and AHC. But I'm sure the LS series won't work here due to clocking limits. :)

jmamma, 2020-05-05:
@frnktrpz please follow @yatli suggestion and uncomment the debug info from the SRAM test. 
Paste output here.

Cheers.

frnktrpz, 2020-05-05:
I’ll do this after work tonight, thanks!

frnktrpz, 2020-05-05:
> @frnktrpz please follow @yatli suggestion and uncomment the debug info from the SRAM test.
> Paste output here.
> 
> Cheers.

> MegaCommand SRAM Test
2200
114
0
0
Failed bank0 zero reading
2200

This is the result. Pressing on the ram seems to make no difference.

jmamma, 2020-05-05:
And different from your first situation.

```
Megank0 mismatch
2200
MegaCommand SRAM Test
2200
SRAM Test Failed bank0 mismatch
2200
```

frnktrpz, 2020-05-06:
Sorry, I was a bit unclear.

First ran the test while pressing on the ram, which resulted in the same outcome. Then ran the test again with lines 57-60 commented out, and that gave the quoted output.

jmamma, 2020-05-06:
Yes. understood.

jmamma, 2020-05-06:
[image]

jmamma, 2020-05-06:
Can you verify VSS and CE2 are 5V.

and VSS is GND

Measure from the top side of the boards on the legs of the chip.

frnktrpz, 2020-05-06:
Around 4.4V on both of those, measuring resistance between VSS and GND on the pcb gives something like 00.2 ohm. 

Have to go back to work in 5 hours so unfortunately won't be able to do any further checks until tomorrow.

jmamma, 2020-05-06:
The SRAM test attempts to write a non zero value to the SRAM and then read it back.
It does this on both banks, and across all available memory addresses, starting at 0x2200.

If the read is zero, then the test fails.

```
2200
114
0
0
```

In your test, it tried to write the value 114 to memory position 0x2200 (the first memory address in the SRAM). When it reads it back it reads 0.

This means the data lines coming from the SRAM chip are all low.

114 in binary =  0b1110010
At least 4 of the data lines should be high. 

--

SRAM Chip enable and source voltage are correct.

Evidence suggests the SRAM chip is not working.

@yatli this is my conclusion. Do you have anything to add?

jmamma, 2020-05-06:
Triple check WE and OE  (write enable, read enable) line from SRAM and arduino double header.

yatli, 2020-05-06:
One more thing to add, also check WE and RESET for 573. If reset is shorted, it will always give zero.

jmamma, 2020-05-06:
output of 573 is only used for address lines. not data. so I dont suspect this is the fault.

frnktrpz, 2020-05-06:
> Triple check WE and OE (write enable, read enable) lines from SRAM and arduino double header.

Cheers. Just got ready to check the traces one more time. Keep getting slightly different results each time I run the test, never gets very far but there's variance every time. Will check back in after this. 
Thanks again, really appreciate the support from you as well as yatli.

frnktrpz, 2020-05-07:
Ok, it seems there is some bad connection going on after all. 
I re-checked the traces, each time a suspect connection came up it was either my poor handling of the probes or a topographical error and double/triple checking verified the connection. 

However one thing that I hadn't yet checked (for which I apologize, it didn't occur to me initially) were the double row headers between the Arduino and the MC board. A couple of the PA pins showed a way higher resistance value than the others. I used compressed air on the female headers and cleaned the pins with some alcohol. 
After reconnecting, the first 3 times the ram test results were the same as before. The fourth time it actually started going, but kept running way beyond 5 minutes and lines like this showed up:
>22
CA78
16
16
16
CA79
86
86
86
CA7A

Stopped the test after 15 minutes or so, reconnected the usb cable, ran the test again and it passed. Just ran it successfully for the second time. I'll run it again a couple of times tomorrow to be sure before re-soldering the oled screen...

jmamma, 2020-05-07:
Great work.

> I'll run it again a couple of times tomorrow to be sure before re-soldering the oled screen

Best to be absolutely certain SRAM is working before moving on to the OLED.

Maybe give those pins a good soak in Isopropyl if you haven't already. 
You could reflow the double header solder joints as well, if you suspect cold solder.

frnktrpz, 2020-05-08:
Reflowed some of the solderpoints on the double header and gave the board one more solid clean today after the ram test failing again this morning. After letting the board dry for a few hours I decided to solder a female header for the oled rather than soldering it on directly as I wasn't sure it was still in working condition after having gone through my desoldering treatment. Reconnected the board, ran the ramtest again, uploaded the firmware and:
[image]

Thanks again for the support jmamma and yatli, troubleshooting this with your help has been a great learning experience for me. Shoutout to micabeza as well for reaching out and offering help. Hope I'll be able to contribute a bit in the future. 
Cheers!

jmamma, 2020-05-09:
Looking good. Well done for persevering.

albadort76, 2021-05-03:
Hi, having a look on your pictures of the thread, I think your C7 electrolytic 10uf capacitor is not in the good orientation.

frnktrpz, 2021-09-19:
> Hi, having a look on your pictures of the thread, I think your C7 electrolytic 10uf capacitor is not in the good orientation.

Haha, you are absolutely right. Didn't notice this until a few weeks ago while building a fresh MC because I wanted one that didn't have the oled screen misaligned because I decided to put it in some headers after my initial mishap. Regardless the MC operated normally for over a year. That's some solid circuit design.     

Sorry for necro'ing but I'm glad to be able to share this :-D

closed 2023-02-24 by frnktrpz