jmamma/MegaCommand_Design #11: Enclosure & knobs, {tall,short}/material recommendation?

yatli, 2020-08-14:
I'd like to print a few official cases and knobs, any recommendations here? I see there are two versions (tall/short).
Also, very cool to see the knob getting modeled! Wonder if I could print rubber-like knobs?

jmamma, 2020-08-14:
You want to print the Tall enclosure. 

_The short version is the same height as the original minicommand case but requires shaving off around 5mm of plastic off the DIN ports._ 

I have flexible filament. I haven't tried printing encoders using it though.

yatli, 2020-08-14:
Another question :p

How do I lock the board to the enclosure? I see 4 through holes on the lid but it's supposed to be locked to the body, not the board?

jmamma, 2020-08-14:
the encoder nuts hold the front of the board in place. and the rectangular strip on back plate presses against the  the MIDI ports.

yatli, 2020-08-14:
Understood.
Placed the "tall" order + 8 encoders. 400 RMB. :D

jmamma, 2020-08-14:
Any idea what quality you printed at?

The encoders knobs might not fit the shaft, depending on the tolerances of the print.
I've had issues here ^

yatli, 2020-08-14:
I guess.. just the normal stuff. 0.1mm tolerance

yatli, 2020-08-14:
Oh and, they offer to embed 4 copper nuts in the body and I happily accepted :p

jmamma, 2020-08-14:
See how you go.

400 RMB sounds like a bargain.

jmamma, 2020-08-14:
Oh and, they offer to embed 4 copper nuts in the body and I happily accepted :p

For the screws?

yatli, 2020-08-14:
Yeah. The cylinders seems to be strong enough to hold them.

jmamma, 2020-08-14:
ha! One of my early designs tried this... The slots I made for the Nuts always ended up being thinner towards the corners, meaning I couldn't align the nuts with the holes correctly.

I gave up on this, and went for Self Tap screws. Which work surprisingly well.

yatli, 2020-08-14:
https://github.com/jmamma/MegaCommand_Design/commit/1339fe99583b68638f9dfa7dfdb793e1ff2d5c07

Emmm... I sent the old lid to them

jmamma, 2020-08-14:
No problem. The fit is dependent on your build. So it might be fine. If not you can just rotate the lid to not use the rise. Or if you still need some stability, you can file down the rise with sand paper.