jmamma/MegaCommand_Design #12: MCL / button and encoder test fail

zerkalo06, 2020-11-24:
Hello Justin and Yatao,

I create this issue after having a conversation with Justin on the elektronaut forum.

At first I had a problem with the sram test, but now after reflowing the ic everything is fine.
I then tried the button and encoder test and I got nothing from the serial monitor, so I checked the SN74LS165AN chips and I got those results (those are in volt of course, even tho I didn't specify it) : 

[image]

I maybe have to check again and de-solder again the male headers as I don't get the same results from those now, 0v all across apart for 47 and +5. Maybe it's just due to the amount of solder, my voltmeter is not that great. Isn't it strange the sram test is still working ?

Here are the pics of my pcb (I replaced one 0,1uf capacitor missing by another ceramic one I had left) : 

[image]
[image]

Thanks for your help, and great great work you doing here !

zerkalo06, 2020-11-24:
a better quality for my results doc : 

[megacommand_1_0_2a12_top_bottom.pdf](https://github.com/jmamma/MegaCommand_Design/files/5590940/megacommand_1_0_2a12_top_bottom.pdf)

jmamma, 2020-11-24:
Better check the basics first.

After compiling MegaCommand Button & Encoder Test Firmware:

Sketch uses 2052 bytes (0%) of program storage space. Maximum is 253952 bytes.
Global variables use 190 bytes (2%) of dynamic memory, leaving 8002 bytes for local variables. Maximum is 8192 bytes.

Serial Monitor should be set to 9600 BAUD.

zerkalo06, 2020-11-25:
I checked. Everything is has intended.

jmamma, 2020-11-25:
[image]

Circuit diagram for reference

jmamma, 2020-11-25:
Background:

The two shift registers are cascaded together to form a 16bit register.
Each bit is representative of the state of switch (encoder button or button). The 16 signals from each of the switches INS[0-15] are connected in parallel to the inputs across both shift registers.

The data is streamed serially to the arduino via PL0. 

Troubleshooting:

1) Can you verify the orientation of R1 and R2. The circular mark on the resistor(s) should be inline with the pin closest to the bottom of the board. The text should be visible from the right side of the board.

zerkalo06, 2020-11-26:
Thanks for the background infos !

If I understand you well R1 and R2 seems to be fine (english is not my first language, I might make mistakes).

[image]

jmamma, 2020-11-26:
yes they look okay.

jmamma, 2020-11-26:
Next step is to verify the continuity of the traces between IC3 and the Arduino.
(Use your multimeter to measure resistance. make sure the traces read close to 0 ohms.)

Please check:
PL0, PL1, PL2

[image]

Please measure from the solder points on the MegaCommand, and on the arduino itself.

zerkalo06, 2020-11-26:
R1 and R2 pins are measuring approx. 0 ohms , so that's good. 

I got PL0 = 0v     PL1 = 4,80v     PL2 = 3,92v

Orientation is good for IC3, and measures seems to match with INS (0v all across, apart for SH/LD (4,82v) and CLK (3,90v))

Also from CLK to PL2 = 0,86v

jmamma, 2020-11-26:
```I got PL0 = 0v PL1 = 4,80v PL2 = 3,92v```

Voltages won't be much use here. You need to check the continuity of the traces on the PCB to check for damage. We need to see that there is a clear connection between IC3 and the Arduino. 

(Use your multimeter to measure resistance. make sure the traces read close to 0 ohms.)

Please check:
PL0, PL1, PL2

zerkalo06, 2020-11-28:
Sorry I didn't understand the first time.

So

PL1 = 0 / PL1 = 0 / PL2 = 0

But there is problems of connection from IC3 to the arduino. None of them are working (results = 1).
There is a white kind of dot on the circuit near IC3, do you think it's flux residue ? I may have to clean it, maybe it's what causing the connection problem.

[image]

yatli, 2020-11-28:
What do you mean by `results = 1`?

yatli, 2020-11-28:
Also, I recommend using the continuity mode (the "beep mode") if your multimeter has one.
Put the two probes on the two target pins and it will beep if the two pins are connected by a path.
The signal path is a thing between two locations in a circuit, so that's why I don't understand your `result=1` thing -- `result=1` against which pin?

Currently you should measure that the correct pairs of locations are connected together:
`PL1 <-> IC3 SH/LD`:
[image]

`PL1 <-> IC2 SH/LD`:
[image]

`PL0 <-> IC3 QH`:
[image]

`PL2 <-> IC2 CLK`:
[image]

`PL2 <-> IC3 CLK`:
[image]

If all of these are connected well, you'll at least get something out of the serial console.

zerkalo06, 2020-11-29:
1 means its disconnected, and 0 that the path is connected, on ohm mode as my multimeter don't have continuity mode. I hope I use it properly. Only did few electronic pedals before this project, working with sound for troubleshooting, so I'm not used to it. Thanks for taking the time, and being patient : ) 

So      

PL1 <-> IC3 = 0     
PL1 <-> IC2 = 0     
PL0 <-> IC3 = 0   
PL2 <-> IC2 = 0     
PL2 <-> IC3 = 0

IC2 SER <-> arduino GND = 0
IC2 A <-> INS 0 = 1
IC2 B <-> INS 1 = 1
IC2 C <-> INS 2 = 1
IC2 D <-> INS 3 = 1
IC2 E <-> INS 4 = 1
IC2 F <-> INS 5 = 1
IC2 G <-> INS 6 = 1
IC2 H <-> INS 7 = 1 
IC2 INH <-> C1 = 0,01

IC2 QH <-> IC3 SER = 0

IC3 SER <-> arduino GND = 0
IC3 A <-> INS 8 = 1
IC3 B <-> INS 9 = 1
IC3 C <-> INS 10 = 1
IC3 D <-> INS 11 = 1
IC3 E <-> INS 12 = 1
IC3 F <-> INS 13 = 1
IC3 G <-> INS 14 = 1
IC3 H <-> INS 15 = 1 
IC3 INH <-> C2 = 0,01

jmamma, 2020-11-29:
```
PL1 <-> IC3 = 0
PL1 <-> IC2 = 0
PL0 <-> IC3 = 0
PL2 <-> IC2 = 0
PL2 <-> IC3 = 0
```

Did you measure these from the solder points on the Arduino Mega?

jmamma, 2020-11-29:
[ ] Verify IC orientation.
[ ] Verify Trace continuity on board.
[ ] Verify Arduino Header continuity. 
[ ] Voltage check.

--

If the above criteria are met, then by process of elimination that suggest either

[ ] short circuit somewhere. 
[ ] faulty IC3

yatli, 2020-11-29:
IC2/IC3 orientation looking good:

[image]

zerkalo06, 2020-11-29:
Yes, I measured from the solder point of the pcb and the arduino.

yatli, 2020-11-29:
```
IC2 SER <-> arduino GND = 0
IC3 SER <-> arduino GND = 0
```

IC2 out and IC3 serial input shorted to ground.

zerkalo06, 2020-11-29:
Sorry, that's just a typo, that's what I meant (I verified, it's still good)

yatli, 2020-11-29:
So, `IC3 SER <-> arduino GND = 1`, correct?

zerkalo06, 2020-11-29:
Yes, that's correct

yatli, 2020-11-29:
```
IC2 A <-> INS 0 = 1
IC2 B <-> INS 1 = 1
IC2 C <-> INS 2 = 1
IC2 D <-> INS 3 = 1
IC2 E <-> INS 4 = 1
IC2 F <-> INS 5 = 1
IC2 G <-> INS 6 = 1
IC2 H <-> INS 7 = 1

IC3 A <-> INS 8 = 1
IC3 B <-> INS 9 = 1
IC3 C <-> INS 10 = 1
IC3 D <-> INS 11 = 1
IC3 E <-> INS 12 = 1
IC3 F <-> INS 13 = 1
IC3 G <-> INS 14 = 1
IC3 H <-> INS 15 = 1
```

These are suspicious. Should be all 0.

zerkalo06, 2020-11-29:
When we're talking about INS1 INS2 etc, we are talking about those ones right ?

[image]

yatli, 2020-11-29:
No, these are arduino pin numbers, which we're not using. Instead we are using the RAW ATMega2560 pin numbers, as shown by @jmamma.

INS pins for the encoders:

[image]

yatli, 2020-11-29:
INS pins for the buttons:

[image]

zerkalo06, 2020-11-29:
Okay my bad. Here we go :

IC2 A <-> INS 0 = 0
IC2 B <-> INS 1 = 0
IC2 C <-> INS 2 = 0
IC2 D <-> INS 3 = 0
IC2 E <-> INS 4 = 0
IC2 F <-> INS 5 = 0
IC2 G <-> INS 6 = 0
IC2 H <-> INS 7 = 0

IC3 A <-> INS 8 = 0
IC3 B <-> INS 9 = 0
IC3 C <-> INS 10 = 0
IC3 D <-> INS 11 = 0
IC3 E <-> INS 12 = 0
IC3 F <-> INS 13 = 0
IC3 G <-> INS 14 = 0
IC3 H <-> INS 15 = 0

zerkalo06, 2020-11-29:
Soooo, I don't understand everything seems good ... 

I tried the test again but still nothing on the serial monitor.

yatli, 2020-11-29:
Measure the resistance between these two:
[image]

Ensure it's neither 0 or infinite, but something around 10K.
Check your multimeter and ensure you're on the right scale.

zerkalo06, 2020-11-29:
ooof thank god I was ready to cast a spell on that circuit.

I get 9.98k for the resistance.

My multimeter is cheap, I planned to get a better one soon : 

[image]

yatli, 2020-11-29:
Okay, resistor array is installed correctly.

yatli, 2020-11-29:
Next, we should isolate the problem of IC, and the buttons/encoders.
Securely connect the Arduino, give it power with USB.
Turn your multimeter to DC voltage 20.

Put the probes to the marked locations, and press the button marked in purple:

[image]

And see if the voltage changes.

zerkalo06, 2020-11-29:
The voltage doesn't change, it stays at 0v. (It does change when in ohms, from 13.40 ohms to 0 ohms)

yatli, 2020-11-29:
Disconnect USB and arduino, check all `INS <-> GND`

zerkalo06, 2020-11-29:
I got : 

0 INS <-> GND = 14.99 ohm
1 INS <-> GND = 14.99 ohm
2 INS <-> GND = 14.99 ohm
3 INS <-> GND = 15 ohm
4 INS <-> GND = 14.99 ohm
5 INS <-> GND = 0 ohm
6 INS <-> GND = 14.99 ohm
7 INS <-> GND = 0 ohm
8 INS <-> GND = 14.99 ohm
9 INS <-> GND = 15 ohm
10 INS <-> GND = 14.99 ohm
11 INS <-> GND = 14.99 ohm
12 INS <-> GND = 15 ohm
13 INS <-> GND = 15 ohm
14 INS <-> GND = 15 ohm
15 INS <-> GND = 15 ohm

jmamma, 2020-11-29:
Did you measure the voltage on R1, and R2 resistor arrays.

The pin closest to the bottom of the board should read 5V with respect to GND on each array.

yatli, 2020-11-29:
> I got :
>
> 0 INS <-> GND = 14.99 ohm
> 1 INS <-> GND = 14.99 ohm
> 2 INS <-> GND = 14.99 ohm
> 3 INS <-> GND = 15 ohm
> 4 INS <-> GND = 14.99 ohm
> 5 INS <-> GND = 0 ohm
> 6 INS <-> GND = 14.99 ohm
> 7 INS <-> GND [...]

These should all read 10K, but now 2 of them read 0 and the others read 15.

zerkalo06, 2020-11-29:
I got nothing from from R1 and R2, 0v. Solder seems alright, maybe I need to try with another multimetre.

yatli, 2020-11-29:
> I got nothing from from R1 and R2, 0v. Solder seems alright, maybe I need to try with another multimetre.

What pins have you attached the probes to?

yatli, 2020-11-29:
If they're all 0v it means that your VCC is shorted to GND...

zerkalo06, 2020-11-29:
GND on the pin under the 0 that appear on the left side of the array, and the other on the last pin on the right. I also tried from GND of the board to the 0 of the array.

The multimeter is on the 20v.

SRAM test is working.

Am I doing it right ?

It might be my multimeter, I'll try to get a new one tomorrow to test that.

yatli, 2020-11-29:
When measuring voltages in this situation you should always put the black probe to GND -- the probe itself isn't called GND -- it's just the negative end of your measurement.

The true GND of the board is marked by the black arrow in my screenshot.
So, just stick the black probe in there, and use the red probe to tap the pins to measure.
You should have 5V on most pins of R1 and R2.

yatli, 2020-11-29:
> SRAM test is working.
> I got nothing from from R1 and R2, 0v

These two contradict each other. So please do the measurement again :)

zerkalo06, 2020-12-08:
Hi ! I receive a new multimeter and did the measures again, oorah.

SRAM test still working.

**For R1 <-> GND  (without usb connected) :**

All pins = 0v

or 

Pin 1 = 5 kΩ
Pin2, 3, 5, 6, 7, 8, 9 = 15 kΩ
Pin4 = 0,2Ω

** R1 <-> GND (with usb connected) :** 

1, 2, 3, 5, 6, 8, 9 = 3,6v
4, 7 = 0v

**R2 <-> GND  (without usb connected) :**

All pins = 0v

or

Pin 1  = 5 kΩ
Pin 2, 3, 4, 5, 6, 7, 8, 9 = 15 kΩ

** R2 <-> GND (with usb connected) :** 

1, 2, 3, 5, 6, 7, 8, 9 = 3,6v
4 = 0v

Regarding the measures for the encoders (voltage without usb connected, GND <-> INS) I got 0v everywhere.

Also :

IC2 H <-> GND = 3,6v to 0v(with button pushed)

zerkalo06, 2020-12-13:
I redid the soldering on R1 and R2 and the encoders. Nothing new with the resistors. 

I then checked the continuity of the encoders and INS1, INS2, INS3, INS4, INS5, INS8, INS9, INS10, INS11, INS12, INS13, INS14, INS15 don't seem to be connected. 

I'll tried to re-solder them but I'm a bit worried that I'll damage the circuit by doing it a forth time.

jmamma, 2020-12-13:
```I then checked the continuity of the encoders and INS1, INS2, INS3, INS4, INS5, INS8, INS9, INS10, INS11, INS12, INS13, INS14, INS15 don't seem to be connected.
```

Highly unlikely. How did you measure this ?

zerkalo06, 2020-12-15:
By attaching black probe to gnd and red probe to each encoders INS pins on the continuity mode, and with the arduino connected. It beeped for only two of them. I had four of them before I re-soldered their pins ...

jmamma, 2020-12-15:
Black probe to GND, and red to INS pin.

Did you push the corresponding button to complete the circuit when testing the continuity ?

jmamma, 2020-12-15:
To recap.

```
Did you measure the voltage on R1, and R2 resistor arrays.

On each array, the pin closest to the bottom of the board should read 5V with respect to GND.
```

Are you still reading 0V on the resistor arrays ? 

Which pin on the mega command circuit board are you connecting the black probe to measure GND?

zerkalo06, 2020-12-16:
Okay thanks, my bad I didn't push the encoder while checking the continuity. They all good.

For R1 and R2 I get only 3,7v on all pins when moving the encoders.

I get GND from the point indicated above by yatlin, near button 2 and marked by a GND sign on the pcb.

jmamma, 2020-12-16:
Okay good to know that. 

On each array, the pin closest to the bottom of the board should read 5V with respect to GND.

This is the supply voltage to the resistor array. Can you verify that it is constant. It should read 5V without rotating encoder or pushing button.

zerkalo06, 2020-12-16:
The supply voltage is constant on the two resistor array but only reading 3,7v or 3,9v depending if the board is powered by a computer or a electrical outlet.

jmamma, 2020-12-16:
Understood.

Can you measure the 5V voltage being supplied by the Arduino Mega.

When viewed from above, the double header has GND, and 5V labelling

zerkalo06, 2020-12-16:
I get 4,8v from both double header.

jmamma, 2020-12-16:
Okay. There's another 5V supply rail that comes off the right most Arduino header in the middle of the board.

I suspect this rail is being dragged down by a failing component. I have a strategy for narrowing it down. 

Just putting together a diagram.

jmamma, 2020-12-16:
[image]

jmamma, 2020-12-16:
All the elements are connected in parallel, so unfortunately each node along that path will measure the same voltage. You measured it at approx 3.7 V.

jmamma, 2020-12-16:
What we know:

SRAM is working. It can run on a supply voltage as low as 2.7V
The Octal latch is working. It must be if SRAM is working. It can run on a low 2V supply voltage.

The shift registers require 4.7V to run. Either one of these could be faulty.

Octo-couplers require 4.5V. Either of these could be faulty.

R1 or R2 should be okay, given your voltage + button test above.

jmamma, 2020-12-16:
So the only option here is to start replacing components connected to the 5V rail. Unfortunately because you soldered up the MIDI circuits, you have two extra ICs that could be to blame.

1) Shift registers.
2) Octal Couplers.
3) Capacitors that are on the 5V rail. <-- easy starting point.

jmamma, 2020-12-16:
So the procedure is:

a) Remove component. 
b) Test the 5V rail. 
c) If 5V rail voltage is measuring 5V you know you've found the problem component.

[1 more reply]

closed 2021-09-16 by zerkalo06