jmamma/MegaCommand_Design #14: Shift1 not working

defalutz, 2021-03-06:
Justin advised me to make an issue here so here goes..

I´ve built about 50+ MCL´s, many thanks to the group buy i organized a while ago, word spread - suddenly i had orders to build to a whole bunch of longing MD users - which was fun and it paid for my own unit plus i like to build and solder stuff, and since this being TH it was a bit easier than i normally build. So most of the units worked great. I think i have had until now maybe 3 with bugs needing to be fixed. Simple stuff, just me being sloppy doing bad solder joints, and one being a memcard not working.

BUT everything has an end! And suddenly i have **two** issues being tricky for me to fix without help, this is one of them.

The issue is with the lower left pushbutton (Shift1); the one that opens the menu. It´s doesn´t do anything at all, no reaction. 

To fix it, I tried these things:

- re-soldering the ICs, the button itself and in general cheking for bad soldering, which i didn´t find
- ran the SRAM test without any trouble
- tried swapping between three different arduino boards, that is not the issue, when i out any of them in my own unit, the one i have used the longest, everything works with no trouble
- ran the button and encoder test = **FAIL** on just the button i have problem with, obviously

What have i done that MIGHT affect the function of the button? 
The only thing that i MIGHT have changed is that instead of the 220 ohm resistors i put 221 ohm ones instead, possibly. I was out of 220´s and figured it shouldn´t matter, not with so little difference.. :)
Also, i have another 9v on/off switch than the EG1218 thats recommended but it works fine in the other machiens that use it, and i have the problem both when running on 9v power and usb so.. shouldn´t be that. Besides it be a little cramped for space and the switch casing are about 1mm or less from three of the SRAM legs i can´t see any problem with it.

I attach some pics of the front and back. Any ideas what to do; replace the button? Or some testpoint that i should check? I´m not super-good at Eagle, i can manage but far from an expert.

If you could give me some pointers i would be very grateful. I´m about to remove the button and test with another one but atm i don´t have one, but waiting for delivery from mouser next week, early to middle

Front:
[image]

Back:
[image]

Thanks in advance! //Marcus

jmamma, 2021-03-07:
Hey Marcus.

You need to get ahold of a multimeter and check the traces connected to the push button.

It's most likely a bad trace, bad solder joint, or short circuit on one of those traces.

[image]

[image]

jmamma, 2021-03-07:
You can check traces by setting the multimeter to measure resistance. And verify that there is continuity from point to point. Continuity means the trace should measure close to 0 ohms.

defalutz, 2021-03-07:
I have one, will check as suggested and teturn with results. Thanks!

jmamma, 2021-03-07:
So it's BUTTON 2. 

The trace to check is INS13.

defalutz, 2021-03-07:
> You can check traces by setting the multimeter to measure resistance. And verify that there is continuity from point to point. Continuity means the trace should measure close to 0 ohms.

I have measured continuity before, my meter beeps when i have it.
The thing about checking traces is good, very useful, thanks!

jmamma, 2021-03-07:
It might not be immediately clear from the circuit diagram, but INS13 is also connected to IC3 (shift register)

Also check the traces on C16.

jmamma, 2021-03-07:
edit: just re-uploaded the traces image because the previous one had some unrelated markings.

defalutz, 2021-03-07:
> edit: just re-uploaded the traces image because the previous one had some unrelated markings.

Great, thanks for the assist, i will test as soon as fatherhood allows. ;)

Off topic
On another note; cant the most skilled with the MCL record a few short vidz that Showcase cool stuff the MCL can do and the Md alone cant? Just to build interest. Food for thought at least, i would IF i had the skills but sadly i have spent My time building. ;)

defalutz, 2021-03-19:
Ok, took a while, but i finally tested with a multimeter. GND for Button 2 was ok. INS13 though did not connect with the corresponding leg joint on IC3.
I used a fully working finished board to compare with just to be sure when testing.
I replaced C24 but problem persisted. When i tried with the working comparison board i got a beep from the button´s INS13 joints all the way to IC3. 

On the error board, when i test between the C24 and IC3 it works, i get a beep, but no beep between button 2 and C24.

I have resoldered the button to make sure but didn´t help. A faulty button might be the issue?
Anything else that i can test?
I am waiting for a mouser shipment, as soon as it arrives i can replace the button2 with a new one, i just hope that this is the issue, but it sounds like it, or what do you think?

Attach a picture with encircled areas of trouble and such.
[image]

jmamma, 2021-03-19:
``` INS13 though did not connect with the corresponding leg joint on IC3.```

The INS13 trace is probably no good.  Try soldering a wire from the shift register directly to the button.

You can't always rely that the PCBs come from the manufacturer in perfect shape.

defalutz, 2021-03-19:
Of course, that might be the issue! I´ve had lots of good boards and no trouble with them thusfar, but it was bound to happen someday. I will try your solution!

Just a follow-up question though.  I drew a white line where i think i´m supposed to do the bypass; but this also bypasses C24 and in the end of INS13 it splits to also connect to the end of R2, the resistor network/matrix or whatever it´s called. What about that? I don´t really know what result that would give if the signal don´t go there also. What do you think?

I could also try do do a small bypass instead, if the trace is just broken from the button2 to the C24 i can solder a small wire between those two instead of a long one and keep the rest of the signal structure intact?

[image]

defalutz, 2021-03-19:
Thanks for the support, i´m not as good at troubleshooting as i wish i would be so - thanks for being available! :)

I´m not worried about damage to the resistor array, just wondering about this; 

* if i choose option #1, with the long cable, i will need to emulate the board design by adding a capacitor like C24 first, then do what i assume is a split of the signal (in the large white circle) that divides the signal so that one goes to the shift register IC3, and the other to the last joint on the R2 array - is this correct?

defalutz, 2021-03-19:
Ah, solved it! Alll i needed to do was to solder a small wire between button 2 and C24 and PRESTO! it worked. :)
Many thanks for your help justin, you and Yateo are the BEST! :)

closed 2021-03-19 by defalutz