jmamma/MegaCommand_Design #15: Too much peering for comfort!

defalutz, 2021-03-20:
Hi,

i have a unit that does an unusual large amount of peering. It still works but especially when i from the grid enter the settings menu, it shows up at every step and takes some additional time that feels too long.
Of all the units i have built before, this is the first i notice this excessive peering. 
It behaves fine besides that afaik.
I have 3.10d installed, as long as X.04 on the MD.
I have run the ram test without issue.
What do you recommend checking?
In the MCL thread at elektronauts Justin recommended taking a look at midi communication on port 1. Anything else that i should be looking for?

Thanks in advance!

defalutz, 2021-03-20:
Ok, done some testing in eagle, and this is the result.

I´ve examined N$14 och N$15, the only ones that have a signal going elsewhere from midi jack 1, U$10

N$14 connects proper to D1 = **OK**,  and then to OK1 (one of the optocouplers 6N137) and leg number 3 there = **OK**

N$15 only goes to R5 and it connects **OK**.

jmamma, 2021-03-21:
sram test?

jmamma, 2021-03-21:
There should be MIDI IN and MIDI OUT test firmwares, you'll need to hookup a MIDI keyboard from memory.

defalutz, 2021-03-21:
> There should be MIDI IN and MIDI OUT test firmwares, you'll need to hookup a MIDI keyboard from memory.

Aha! That's good, i hope it will help. 
The problem is so small i almost skipped posting it here, but i Will see if i can shoot a video and show to you just to clear it Up a bit.

What do you mean to hookup a KB "from memory"?

Best,
Marcus

yatli, 2021-03-21:
guess "from memory" is short for "iirc" :D

yatli, 2021-03-21:
what I did back then is to loopback the midi ports and verify the data.

defalutz, 2021-03-22:
Ok, i checked the midi in with a keyboard and out with midi monitor in bitwig; seems like a get a proper response, or what do you think? I get no breaks, nothing strange happening and the values seem correct. Besides that midi monitor in bitgi doesn´t say note "off" it gives two "on " on the same note.. but it seems to be normal for it. Attached some screenshots to verify!

[image]
[image]
[image]
[image]

jmamma, 2021-03-22:
note_on when velocity = 0, is note off.  so the two note_on messages are okay.

jmamma, 2021-03-22:
Can you try adjusting turbo settings on the faulty port. 
Start at 2x, and work your way up. 

See if peering is okay at lower speed?

Also, have you tried another arduino ?

defalutz, 2021-03-22:
I will check that. But i don´t know which port is faulty, but i will try with in 1 first as you say.

Yes, i have tried with at least three different arduino boards so i´m pretty sure it works. Also even tried with a clone board. Same same.
But the things i can test you mentioned is a good idea, i´ll be back with results. Thank you. :)

jmamma, 2021-03-22:
What image is being displayed on the Peering screen. Is it the Machinedrum?

If not, in MCL global settings menu. check Settings -> MIDI -> Device 2 

Set this to GENER for General MIDI.

defalutz, 2021-03-22:
Eh... do you mean MCL can show a different image depending on the device connected when it says "peering"? Not sure actually but i will check now!

defalutz, 2021-03-22:
Well, i checked what you suggested; i had elektron as device two, but since i use port two in to get clock from my DAW it makes more sense to set it to generic.
I don´t know what´s changed but suddenly i don´t get the peering screen almost at all anymore. Tried changing turbo settings but that didn´t change anything so now i leave it at 8x. 
But it makes more sense to lease the p2 at generic.
Dont know why but suddenly i have no errors anymore so i guess the unit it ok?
All i have dome besides error searching and what you suggested was re-upload the firmware again to it, but nothing new.

But anyway, i think i will send it to the customer and be done with it since it now shows no bugs at all.

And if he has bugs i have the thread with suggestions. 
Thanks for the assistance, at least i have learned something. :)

jmamma, 2021-03-22:
I noticed setting Device 2: to Elekt will cause the MC to try and peer even if there is no MIDI activity.
That might be what you were seeing

defalutz, 2021-03-22:
Yes, that might be the case. Though i´m pretty sure i have done it before without trouble but i can be wrong. But it seems to work now so all good. :)

closed 2021-03-22 by defalutz