jmamma/MegaCommand_Design #16: Leftmost encoder SW1 not working, INS0 not connecting with res.matrix R2

defalutz, 2021-04-23:
I had trouble with odd behaviour from the module, after looking over it a few times for the most obvious faults i found the problem.

The inner (R2) of the two resistor matrices was oriented the wrong way around, so it had the text pointing into center of the board instead of outwards as the outer one was/is.

I had a little trouble removing it so i worried (not without cause, it seems) that it would come with a large bill in the end. 

Result:

The first encoder (SW1 in eagle), the leftmost, does not work at all. I cannot more the cursor or do anything, change values and so on.

After testing the connectors, one after another:

ins8 har conn to end destination = OK
ins1 as above = OK
ins0 connects to the next to last point, which is on IC2 but not after that!

The next and last connection for INS0 is with the the third point on R2. This is not happening.

I have tried with the multimeter on the bottom side and on the top side between the yellow "sausage" of the matrix itself - both tries without success.

I must have f*cked something up when desoldering it or it didnt work from the start, who knows?

I am no master at desoldering so it walways gives me the shivers when i have multiple points i have to desolder to remove something. It´s 50/50 to succeed.
I forgot that i have a hot air soldering/desoldering station i could have used instead but now it´s probably too late. Or that do you think, should i try again ?

I´m not sure it will work even if i do it better next time because there might be some broken trace or something that is the cause. Of maybe one of the other solder joints to the matrix is fudged, who knows? 

I could try it of course and will do if i cant figure out something else.

But i wonder uf there is anything else i can do instead of drastic things. Or any other drastic things i can try? How can i get INS0 to connect even if the solder joint is gone? Can i solder a small cable or jumper from the topside closest to the "sausage" to somewhere else? It´s not clear to me where it goes next or how the matrix even works?

Any ideas? Thanks!

[image]

jmamma, 2021-04-24:
The board can be easily damaged during a desolder attempt. 

If the trace is damaged you'll have to repair it using a jump wire like before.

If the pad is damaged, you'll have to solder a wire directly to the pin on the resistor. 
Another solution would be to solder a 10K resistor between INS0 and +5V . This would bypass the resistor array.

defalutz, 2021-04-24:
> The board can be easily damaged during a desolder attempt.
>
> If the trace is damaged you'll have to repair it using a jump wire like before.
>
> If the pad is damaged, you'll have to solder a wire [...]

I can do both of the two later solutions without problem, but it doesn´t affect anything else by doing that?
You shouldn´t have recommended it if it had of course, but.. just asking.. 

Ok, i will try to do the jumper first; just to make sure, you mean i should put a jumper like this below? Looks crappy i know but at least the angle makes it possible to do.

[image]

defalutz, 2021-04-24:
Alright, i´ve read up a bit on the resistor NETWORKS (not matrices)... doh.. And the solution suggested worked just fine, thanks. I soldered a thin cable and now the unit works great - plus i have learned a lot. And next time i will use other techniques when having to de-solder components like this.
Thanks. Issue closed. :)

closed 2021-04-24 by defalutz