jmamma/MegaCommand_Design #17: PROG NOT COMPATIBLE

albadort76, 2021-05-03:
Hi,
 All was fine with my MCL 3.10d until i the SD card has been broken in use. Can't repair it with disk utility.
Since i have a new SD card 16go FAT 32 formated, i try to create empty project just to check if it's working (without the MD connected to). Some of the project created are working some others are "NOT COMPATIBLE" project. For this kind of files, during the creation process, MCL shut down and restart before the end of the process.
I think i have a problem around the SD card somewhere, before this all was working good...

albadort76, 2021-05-03:
After several tests it seems to work again! Connection issue?

jmamma, 2021-05-05:
Sorry. I thought I replied. The issue could be a connection issue with the SDCard holder. Maybe a bad solder joint?

Depending on your soldering skills, and soldering iron, it is possible to reflow the SDCard holder pads with the screen still in place.

albadort76, 2021-05-05:
No problem! 
 Hopefully got a good soldering iron with a very fine head so that's exactly what i have done everything's seems to be fine since with the new SD card.
  Hard to find them under 32go now, maybe some day all this space will be useful (WAV!! :)) ? I'm working on a new track i let you know. Cheers

jmamma, 2021-06-17:
@albadort76 have you had any additional problems with the SD card since your last post?

albadort76, 2021-06-17:
No more problem with the new sd card after reflewing solder joints.

jmamma, 2021-06-17:
Good to hear. I gave the SD Card code a once over. Found some scenarios where reads could be unreliable. So that will be improved in next version.