jmamma/MegaCommand_Design #18: MIDI send/receive firmware test fail

jabarbanel, 2021-07-16:
Hi Justin-

I have moved this issue over from the Elektronauts forum. I posted there last week about having issues with the MIDI ins and outs of my MegaCommand build. 

Summary - I've been building my MegaCommand as per the instructions on GitHub. Passed the SRAM and button/encoder tests and have now hit a wall at the MIDI out/in firmware tests.

My setup: I have a MOTU MIDI Express XT connected to my PC and Ableton. I’m using the front panel MIDI ports (channel 8) to send/receive. With my Machinedrum connected, I can play sounds from the machinedrum with my keyboard, and when I play a pattern on the machinedrum I see MIDI note info lighting up on the MIDI interface (as well as in Ableton). So I confirmed that the MIDI setup is working as it should.

Here is the latest update on what I have done as far as troubleshooting: I am only working on the MIDI out circuit since it is so simple - but I did confirm the optoisolators are installed correctly for the input circuits.

1. Confirmed the Arduino is functioning properly. I was able to recreate the MIDI out circuit on a breadboard to confirm my Arduino is working:
[image]
With this setup I ran the MegaCommand MIDI-out test firmware and confirmed note data being received by my MIDI interface.
(side note - I was working with an older cheap arduino clone that had the rx1/tx1 pins either mislabeled or miswired, so I bought another clone and confirmed the pins are correct and functional)

2. Checked continuity between all the points of the MIDI send/receive circuitry on the pcb between from the arduino headers all the way to the MIDI connectors - everything looks good.

3. Cleaned the PCB with 99% Isopropyl Alcohol

4.  Checked another PCB from my order (I had the minimum of 5 made) by connecting only the MIDI out connections and resistors for TX1 output (see picture) - and WAS able to produce MIDI out notes that I confirmed on my interface:
[image]

5. Visual inspection of solder joints (looks OK to my untrained eyes)

6. Used a multimeter to confirm the resistors are 220Ohm - check

I'm really confused about this! I really would appreciate any insight / guidance on how to proceed at this point. Also here are pics of the build so far for reference.

[image]
[image]

Thanks for your time Justin, I really appreciate it.
-Jonathan

jmamma, 2021-07-17:
Jonathan your MIDI jacks are installed on the wrong side of the PCB!!

jmamma, 2021-07-17:
Please see the reference photography.

https://github.com/jmamma/MegaCommand_Design/tree/master/1_0_2/photos

jabarbanel, 2021-07-17:
omg. I wish there was a word for feeling simultaneously elated and mortified at the same time. Elated to have the solution be so simple and obvious. Mortified that I made such a bonehead rookie mistake.

Anyway thanks for the response! I have some fun desoldering work ahead - just ordered a fresh roll of desoldering braid because my solder sucker isn't removing enough to remove these jacks.

I'll post a confirmation once I get these jacks installed properly just to have some closure.

Thanks again Justin. I'll just show myself out... 
<html>
<body>

:flushed:
--

<br class="Apple-interchange-newline">
</body>
</html>