jmamma/MegaCommand_Design #20: Peering issue

sc0rdatura, 2021-09-17:
[image]
[image]

Hi :),

I’m just flagging up that my megacommand keeps getting stuck on peering when it’s connected to the MD mk2. This is despite all tests passing and the MD and MegaCommand both being updated to the latest firmwares. I’ve tried a few different MIDI cables but it’s the same still, and the ports all pass the MIDI tests.

Thanks for your help!

jmamma, 2021-09-17:
What type of solder did you use? Post a photo of the label ?

You have a lot of flux residue covering the pads of the MIDI ports.

sc0rdatura, 2021-09-17:
It’s 60/40 Sn/Pb 5C ersin flux. Don’t have the label on the solder anymore but I know that’s what it is.

jmamma, 2021-09-17:
Ersin 5C solder apparently uses a rosin flux, so it should be non-conductive. 

https://rmscomponents.com.au/wp-content/uploads/2018/02/Multicore-Ersin-Fluxes.pdf

sc0rdatura, 2021-09-18:
I have inspected/reflowed the joints around the MIDI jacks but sadly the problem persists. The flux doesn’t seem to be the issue.

jmamma, 2021-09-18:
[image]

Can you reflow the joints highlighted in GREEN. This is the header linking the MIDI circuitry to the Arduino. Corrosion on the pins and or bad header on the Arduino could also be a source of intermittent connection issue.

Please clean your board of any left over solder blobs located in YELLOW and elsewhere on your board.

sc0rdatura, 2021-09-18:
[image]
[image]

I have inspected all the joints on the board, reflowed (especially those in green) and removed flux for good measure. Solder blobs have also been removed. The issues sadly still persists. The peering sequence follows a pattern, whereby as soon as the MIDI in of the MegaCommand is connected to the MD it says “peering”. The progress bar moves for 10 secs or so then freezes, and then briefly flashes to the grid mode for 0.5 seconds before returning to the peering screen.

jmamma, 2021-09-18:
What happens if you set Config -> MIDI -> TURBO 1: 1x

jmamma, 2021-09-18:
I've uploaded the MIDI circuitry for reference. 

[image]

I suspect the pullup resistors R11 and R12 highlighted in yellow are of incorrect value. They look different to mine.
Can you check what part you used? They should be 1Kohm

[image]

sc0rdatura, 2021-09-18:
Have just tried, sadly no change still.

sc0rdatura, 2021-09-18:
Have tried both the 1x midi setting and confirmed the resistor values are at 1k by Color code and measurement. Still getting the same result.

jmamma, 2021-09-18:
You can check supply voltage to the 6n137s with a multimeter. Supply voltage should be 5V.

sc0rdatura, 2021-09-18:
Can confirm the couplers are getting the correct voltages.

jmamma, 2021-09-18:
Do you have another Arduino you could test with ?

sc0rdatura, 2021-09-18:
Sadly I don’t have a spare arduino mega, but is the arduino a possibility even if all of the build firmware tests pass?

jmamma, 2021-09-18:
yes. could be a bad connector or dodgy board. 

The MIDI firmware test is only testing note data. not large streams of SYSEX which are exchanged during peering.

jmamma, 2021-09-18:
With the Arduino removed. You could use your multimeter to measure resistance between TXD1 and PIN5 of the MIDI OUT 1

It should read 220 ohms.

sc0rdatura, 2021-09-18:
Yup that all looks good as well. Just did some testing around the midi jacks/couplers with the multimeter and your schematic and everything seems in order.

jmamma, 2021-09-18:
Running out of potential candidates.

1) Can you measure GND on the 6n137s against GND on the double pin header. It should read 0V

2) Are you powering via USB or DC? you could try the other. 

3) Do you have an Analog 4 or MNM ? If so, you could try connecting one of those on PORT2 and see if peering is working  on PORT2.

jmamma, 2021-09-19:
Probably should have checked the most obvious first.

You connected 2 MIDI cables between the MD and MCL ?

Peering requires both MIDI IN and MIDI OUT connection to the MD.

sc0rdatura, 2021-09-19:
I’ve been trying both with USB and DC, to see if it would make any difference, but alas not.

I don’t have either to test port 2 sadly.

Yes I’ve been connecting MIDI both in and out. The peering only starts showing when I plug in the out of MD to in of MCL.

frnktrpz, 2021-09-19:
Hope it's ok for me to comment here but if I were you I'd give the whole board another reflowing, still a bunch of dodgy looking joints imho. And after that give it a really thorough clean with isoprop and a toothbrush, clean it til it's spotless and then go over it with a can of compressed air, blow it under the ic's while holding the board sideways and into the headers with the board upside down. 
I had some issues last year and everything was resolved after really properly cleaning the board.

defalutz, 2021-11-10:
Hey, i have Some peering issues, this One just won't start peeering what ever i do. I have reflowed as much as felt reaśonable at the moment. Cables are fine since i used Them yesterday to this purpose. I have cleaned the board reasonably thorough. 

Heres a few pics, can you see anything out of the ordinarie?
[image]
[image]

jmamma, 2021-11-10:
[image]

The silk screen seems to be missing near some of the solder pads? 

If solder bleeds from a pad to nearby exposed copper, it could result in a short.

defalutz, 2021-11-11:
Yes, i noticed that myself when i posted, i saw in eagle that two traces from the first midi port go to the right and i'll check if One of Them is Close to this exposed copper and if there is a connection where it should not be. Hopefullly it's a matter of cutting the trace!
Thx, Will let you know! :)

jmamma, 2021-11-11:
You shouldnt need to cut any traces, just make sure you dont have any shorts by solder sticking to copper where it shouldnt be.

defalutz, 2021-11-11:
Ok, got it. Unfortunately it didnt make a difference.
Back to experimenting. Peering, Short break, peering and so on. I'll check in some of the other ideas in the thread too.

defalutz, 2021-12-04:
Hey guys; i have actually found out some interesting things. On the main midi ports, peering just restarts all the time. On the secondary midi ports, if i set them to elektron and to receive midi, peering works. Though i havent found nything that differs from them right now. Any ideas? Thanks. Pics coming soon! Anything except the 6n137 i should check voltage on - or any ideas on how to compare the ports properly?

frnktrpz, 2021-12-05:
Don't mean to come off harsh here but I'd prioritize improving your soldering and keeping your boards clean before chasing down errors. From the pictures you posted there seem to be quite a few soldering points across the board that need reflowing/reworking. Maybe it's the lighting in the pictures but it looks like there might be a couple of cold joints, insufficient wetting and with double sided boards you generally want to be aiming to get a nice small concave mound on the component side as well as the solder side. (https://www.manualslib.com/manual/2085266/Thonk-Eurorack-Diy-Kit.html?page=2)
Also there seems to be a rather large amount of debris (dust/specks of solder?) on the board and components. And what's up with all the bits of missing soldering mask?

defalutz, 2021-12-05:
No, dont worry. I appreciate the input. And it´s true. The issues you describe is because of 
1. stress, a lot to do and too little time, so unfortunately in some cases i might have been sloppy soldering
2. my usual solder ran out, i had to use one with different characteristics, and at the same time i started using new tips for my soldering iron and together they behaved not as i am used to - it got a lot harder to get the joints to look the proper way so it´s entirely possible it might be a few cold joints and so on and so forth.
3. I got a batch of low quality PCBs  at the same time as the above things, and where i had NO problems before, suddenly with this batch i started having isues all over.
I usually clean up the build so unfortunately you got pics taken from before the clean-up.

The solder mask was very sensitive (again, this high level of sensitivity i´m not used to) so it came off during certain manouvres or if i had to desolder a component and replace if with another.

Again, thanks for the hints. I realize i have to find a better combination of solder and tip for the iron, a pair that work better together. Also, i need to clean up the board before and after better, to make the solder stick better for example..

I´ve made a sh*tload of working MCL units before so this has baffled me a bit, but i´ll cope with it. At the moment i´m quite annoyed with the boards used for the three units i´ve had trouble with, so i will send them into a black hole after removing all useful components. Then they will end up on different PCB´s that i´ve had zero trouble with so far..