jmamma/MegaCommand_Design #22: SRAM Addressing issue?

possan, 2022-05-22:
Hello, i was handed a "broken" MegaCommand device and just started to do some basic reflowing of the solder joints.

After running a few of the testing firmwares successfully, it finally failed on the SRAM one, i modified it a bit to print out what it reads from which address and that's when i saw an interesting pattern, it seems like the two lowest address bits are high which means it reads a slightly higher address than expected, every fourth read is correct.

```
Address 22F7 (10001011110111)  correct F7 (11110111) 
Address 22F8 (10001011111000)  read FB (11111011)  expected F8 (11111000) 
Address 22F9 (10001011111001)  read FB (11111011)  expected F9 (11111001) 
Address 22FA (10001011111010)  read FB (11111011)  expected FA (11111010) 
Address 22FB (10001011111011)  correct FB (11111011) 
Address 22FC (10001011111100)  read FF (11111111)  expected FC (11111100) 
Address 22FD (10001011111101)  read FF (11111111)  expected FD (11111101) 
Address 22FE (10001011111110)  read FF (11111111)  expected FE (11111110) 
Address 22FF (10001011111111)  correct FF (11111111) 
Address 2300 (10001100000000)  read 3 (11)  expected 0 (0) 
Address 2301 (10001100000001)  read 3 (11)  expected 1 (1) 
Address 2302 (10001100000010)  read 3 (11)  expected 2 (10) 
Address 2303 (10001100000011)  correct 3 (11) 
Address 2304 (10001100000100)  read 7 (111)  expected 4 (100) 
Address 2305 (10001100000101)  read 7 (111)  expected 5 (101) 
Address 2306 (10001100000110)  read 7 (111)  expected 6 (110) 
Address 2307 (10001100000111)  correct 7 (111) 
```

I checked the traces between the Arduino data pins and the sram and latch, and they seem correct and no crosstalk.

I'm suspecting that the latch is broken, but it seems weird that only a few bits of the latch is broken, and i haven't been able to run it through a scope yet so i'm not sure if they're actually high or low.

I'm just curious if anyone have seen something like this before?

jmamma, 2022-05-23:
It sounds like you're pretty close to finding the problem.

This is the reference diagram for the SRAM from the Atmega 2560 datasheet.

[image]

With the aid of the Latch, AD7:0 is providing both high Address lines and Data, if the problem was on the AVR side you would expect to see stuck bits in the data as well.

Are all the reads ending in 0b11 ??

possan, 2022-05-24:
Not sure, it seems like it, i guess i could try that by just writing the two lowest bits as zero and see what it returns, i'll do some more digging in a few days..